gw-vis/userapps #109: Release sensor monitor

MiyoKouseki, 2020-09-10:
[from here](https://www.dropbox.com/s/hoadb73fdyah02c/Post-O3%20task.xlsx?dl=0)

gw-vis/kagra-vis#16 : How do we connect the relase sensor to the binary input.

MiyoKouseki, 2020-09-30:
@Ryutaro-T @vis-ikeda これってインストール時しか使わないやつですか？池田さんにお願いしてデジタルシステムでモニターできるようにする必要がありますか？

Ryutaro-T, 2020-09-30:
常にモニターできるようにする必要があります。

vis-ikeda, 2020-10-15:
@Ryutaro-T @MiyoKouseki いつまでに必要か含めて具体的な内容をどこかで教えてください。

MiyoKouseki, 2020-11-05:
リリースセンサーをどうDGSに繋ぐかを別のIssueで議論し始めているのですが、これはどうすればいいでしょうか。手作りはしたくないので、AELの方にお願いしたいのですが、これがどれぐらいの優先度でいつまでに欲しいのか明らかにしておきたいです。

MiyoKouseki, 2020-11-05:
EYのインストール作業で用意できれば良いが、ローカルで回路ボックスに繋いで確認する方が便利なので、特にこの時までに必要ということではない。

インストール後で良いと思う。来年4月とかで。

MiyoKouseki, 2021-03-12:
https://dacgit.icrr.u-tokyo.ac.jp/issues/546?issue_count=37&issue_position=2&next_issue_id=541&prev_issue_id=547

に移動

closed 2021-03-12 by MiyoKouseki