gw-vis/userapps #110: Enable IP stepper motor limit switches which only require software additions (as explained in JGW-E2011534).

MiyoKouseki, 2020-09-10:
[from here ](https://paper.dropbox.com/doc/Plan-for-the-long-brake-towards-O3c--A6BHN4gn~Ey5XkTLymY8BCZaAg-KqexeqMaCA7DITFGxA5ih)

## Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

## To Do 
List up the To Do.

MiyoKouseki, 2020-09-15:
Dear Miyo-san and Fabian

I am sending a list of commands related to the limit switch. Please find
the attached files. (Pdf file is firmware manual of motor driver.)

The important thing is not to confuse right and left in wiring the
switch. For example, if the limit switch located at the end of a left
turn command to advance is wired to the "right, the limit switch does
not play a role. I heard there are parameters which can reverse the
definition of direction of rotation in KAGRA control system. I worry
about if applying them to limit switches will cause them to run out of
control.

The stepper motor for testing is now in Mitaka. If needed, I will send
it to Kamioka.

Please let me know if you have any questions.

Best regards,

Sato

```
Command list for limit swich

# ------------------------------------------------------------------------------
# <motor No.> のリミットスイッチを有効にする
# Enable the limit switch of <motor No. >.

SAP 12, <motor No.>, 0   # Right SW enable
SAP 13, <motor No.>, 0   # Left SW enable

----- binary ------
command No.: 05h
Type       : 0ch (0dh for Left SW)
Motor/Bank : 00h - 05h
Value      : 00h

# ------------------------------------------------------------------------------
# リミットスイッチ有効状態を不揮発性メモリに記憶
# Stores the limit switch enablement status in nonvolatile memory

（注意）必ず「リミットスイッチ有効」にセットされている時に実行する
(Note) Be sure to execute the operation when the motor driver is set to "Limit switch enabled".

STAP 12, <motor No.>   # Right SW
STAP 13, <motor No.>   # Left SW

----- binary ------
command No.: 07h
Type       : 0ch (0dh for Left SW)
Motor/Bank : 00h - 05h
Value      : 00h

# ------------------------------------------------------------------------------
# モーター駆動（現在位置から指定されたmicrostep数駆動する）
# Motor Drive (drives a specified number of microsteps from the current position)

（注意）このコマンドはモーターが最終位置に達するのを待つことなく、さらにコマンドを送ることができる。→ MSTコマンドで中断可能。
(Note)This command can sends a further command without waiting for the motor to reach its final position. So you can abort by using the MST command. 

MVP REL, <motor>, <offset>

----- binary ------
command No.: 04h
Type       : 01h
Motor/Bank : 00h - 05h
Value      : (4byte, 有効範囲は 32bitまで、有効範囲を超えると逆回転する可能性あり)
4 bytes, effective range is up to 32 bits, but there is a possibility of reverse rotation if it exceeds the effective range.

# ------------------------------------------------------------------------------
# モーター停止
# Motor stops

MST <Axis>

----- binary ------
command No.: 03h
Type       : 00h
Motor/Bank : 00h - 05h
Value      : 00h

# ------------------------------------------------------------------------------
# リミットスイッチの開閉状態を取得
# Get the open and closed status of the limit switch.

GAP 10, <motor No.>   # Get Right SW state (0 or 1)
GAP 11, <motor No.>   # Get Left SW state (0 or 1)

----- binary ------
command No.: 06h
Type       : 0ah (0bh for Left SW)
Motor/Bank : 00h - 05h
Value      : 00h

# ------------------------------------------------------------------------------
# ------------------------------------------------------------------------------

Command Example (for Motor No.0)

SAP 12, 0, 0
SAP 13, 0, 0
STAP 12, 0
STAP 13, 0

MVP REL, 0, +90000

<If the limit switch is activated and the motor stops, need to execute the "MST".>

MST 0
# ------------------------------------------------------------------------------

```

vis-ikeda, 2020-09-15:
手元にあるYaw駆動機構試作品のLとRのSwが逆に付いている気がします。
[正常]
　モーター側 = L = Pin3-8 = GAP 11(Left)が1
　モーターから遠い側 = R = Pin4-9 = GAP10(Right)が1
[現状]
　モーターから遠い側 = L = Pin3-8 = GAP 11(Left)が1
　モーター側 = R = Pin4-9 = GAP10(Right)が1
になっておりました。
写真ではわかりにくいですがこの状態でGAP11(Left)側のデータが1でした。

接続が逆ではないかと考えていますので要確認。

[image]

[image]
[image]

NaohisaSato, 2020-09-15:
池田さま
テスト装置の配線は、配線手順書 https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11520 を作る前に組み立てました。LとRが逆になっているかもしれません。

vis-ikeda, 2020-09-15:
LとRを配線手順書に合わせて配線を入れ替えました。

Reference Search Modes:3でL/RのSWが検知できることが確認できました
のでやはり逆に付いていたようです。
SWは正常に検知できたので作業を進めます。

vis-ikeda, 2020-09-16:
・stepmotor理解の為、クラス図の作成
[uml_class.drawio.svg.pdf](https://github.com/gw-vis/userapps/files/5229879/uml_class.drawio.svg.pdf)

vis-ikeda, 2020-09-16:
・Trinamic_control6110.pyのgetRightLimitSwitch()/getLeftLimitSwitch()を使ってLimitSwitchの状態を取得
###  [IPアドレスの変更]
ブラウザで192.168.127.100へアクセスして変更
- 初期状態
**Network Settings**
IP address: 192.168.127.100
NetMask:255.255.255.0
Gateway:空欄
**Operation Settings**
Operation Mode: RealCOM
- 変更後の設定
**Network Settings**
IP address: 10.68.150.63 [(空きのアドレスで試験)](http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/DGS/Projects/StepperMotor#Driver_List)
NetMask:255.255.0.0
Gateway:10.68.10.1 [k1gate]
**Operation Settings**
Operation Mode: Tcp Server

### [テストコード作成]
- とりあえずコピーして試験
$ cp Trinamic_control6110.py t6110.py
t16110.pyへ追加
```
if __name__ == '__main__':
    tc = Trinamic_control6110()
#    tc.connectTCP('130.235.95.232', 4001)
    **tc.connectTCP('10.68.150.63', 4001)
    print "getRightLimitSwitch",tc.getRightLimitSwitch()
    print "getLeftLimitSwitch",tc.getLeftLimitSwitch()
    tc.reconnect()**

```
### [試験手順]
・LimitSwitchを手袋をしてL/RそれぞれのON／OFF状態でログを確認

### [試験結果]
・L/R共にOFF
(base) controls@k1ctr7:/opt/rtcds/userapps/release/vis/common/scripts/epics-motor-control/stepmotor/steppingmotor$ python2 t6110.py
getRightLimitSwitch 0
getLeftLimitSwitch 0
Reconnecting...
Testing connection:
...ok

・L側:ON、R側:OFF
(base) controls@k1ctr7:/opt/rtcds/userapps/release/vis/common/scripts/epics-motor-control/stepmotor/steppingmotor$ python2 t6110.py
getRightLimitSwitch 0
getLeftLimitSwitch 1
Reconnecting...
Testing connection:
...ok

・L/R共にOFF
(base) controls@k1ctr7:/opt/rtcds/userapps/release/vis/common/scripts/epics-motor-control/stepmotor/steppingmotor$ python2 t6110.py
getRightLimitSwitch 0
getLeftLimitSwitch 0
Reconnecting...
Testing connection:
...ok

・L側:OFF、R側:ON
(base) controls@k1ctr7:/opt/rtcds/userapps/release/vis/common/scripts/epics-motor-control/stepmotor/steppingmotor$ python2 t6110.py
getRightLimitSwitch 1
getLeftLimitSwitch 0
Reconnecting...
Testing connection:
...ok

vis-ikeda, 2020-09-23:
**Input** Link

**development documents Documentation**
[CAS: Channel Access Server Library](https://epics.anl.gov/extensions/cas/index.php)
[tmcm-6110 Documentation](https://www.trinamic.com/products/modules/details/tmcm-6110/)

**memo.**
 ・tmcm-6110のBlock Diagram / Pinoutからreference switchs(ここではLimit Switchを含む)は、TMCM-6110に内包するMotion Controller(TMD429)に直結している。
TMD429は単独でSwitchを検知することができる。
Limit Switchの制御を禁止しなければ自動的にモーターを停止する。
　その為、コマンドでLimit Switchを制御するのはsatoさんの記述の
```
# <motor No.> のリミットスイッチを有効にする
# Enable the limit switch of <motor No. >.

SAP 12, <motor No.>, 0   # Right SW enable
SAP 13, <motor No.>, 0   # Left SW enable
```
のみ。後は勝手にTMD429が停止させる。

上記から考えると、LimitSwitchの制御は行わなくても良いので
上位で1sec程度で状態をポーリングしてユーザーに通知する方法を検討する。

vis-ikeda, 2020-09-25:
[確認手順]
CH.0から5までモーターを接続して、
caget K1:STEPPER-TEST_TEST_x_LSWITCH
caget K1:STEPPER-TEST_TEST_x_RSWITCH
でL/R limit Switchが取得できることを確認
caput K1:STEPPER-TEST_TEST_x_STEP 1000000でRSWITCH:1, LSWITCH:0
caput K1:STEPPER-TEST_TEST_x_STEP -1000000でRSWITCH:0, LSWITCH:1

[確認結果]
・CH0,1,2,4,5については正常にLimitSwitchの状態が取得できた。

[問題点]
・CH３に接続するとモーターが駆動しない、L/RLimitSwitchは動作する。
　→ 原因不明。モーターに通電がされていなさそうでした。
・1/5回L側でLimitSwitchがON状態であてついた時、モーターを反転させても動かなかった。
　→ 原因不明。別issueで管理。（サンプルは試作品でかつ一度LimitSwitchが逆についた状態で当てつけている為か？）

vis-ikeda, 2020-09-28:
**MEDM画面へlimit switchの追加してMDEMから状態が確認できることを確認**
[対応内容]
下記ファイル編集してLimitSwitchの状態を追加
https://github.com/gw-vis/pcas-controller/tree/master/medm/stepmotor
CUST_STEP.adl
CUST_IP.adl：ここへ追加してグループ化したものを下記へコピー
CUST_IP_BS.adl：IP_BSのみ名称を固定化した。
CUST_IP_SR.adl
CUST_IP_TM.adl

**[試験内容]**
CUST_STEP.adl:
　・試験環境にて./step_start.py TEST_GASを実行してEPICSチャンネルにLimitSwitchが流れるようにする 
　・試験環境(unittest.adl)からMEDMを表示して表示を確認
　　表示は自動更新ではなくUpdateを押下してLimitSwitchの状態を確認

CUST_IP_BS.adl：
　・IP_BSのみ名称を固定化しているのでLimitiSwitchの確認はできない。

CUST_IP_SR.adl
CUST_IP_TM.adl

　試験環境が整わない為、当て付けの試験はせず
　LimitiSwitchのチャンネルが取得できていることを確認
　チャンネルが取得できていないとLimitSwitchの表示が白になる。正常時は緑、赤表示
　LimitSwitch端子が開放（未接続）状態であればL/R共に不定（現状両方とも赤）

vis-ikeda, 2020-09-28:
[試験結果]
・CUST_STEP.adl
　・STEPへ1000000を設定してモーター駆動→停止後、
　　Updateを押下してLimitSwitchの状態を確認
　　結果：Left:緑、Right:赤であることを確認　

　・STEPへ-1000000を設定してモーター駆動→停止後、
　　Updateを押下してLimitSwitchの状態を確認
　　結果：Left:赤、Right:緑であることを確認　

・CUST_IP_BS.adl
　MEDMが開けること、現在のBS_PIの状態が表示されること
　Limit Switchが白のまま（チャンネルが未登録）であることを確認

・CUST_IP_SR.adl
・CUST_IP_TM.adl
　MEDMが開けること、Limit Switchが白にならないことを確認

vis-ikeda, 2020-09-28:
何度か試しているとscripts側が死ぬことがあり要調査

raceback (most recent call last):
  File "/kagra/apps/pcaspy/lib/python2.7/site-packages/pcaspy/driver.py", line 450, in getValue
    newValue = driver.read(self.info.reason)
  File "steppingmotor/pcasip.py", line 210, in read
    value = self.driver.getLeftLimitSwitch(motorAddrC)
  File "steppingmotor/Trinamic_control6110.py", line 528, in getLeftLimitSwitch
    data = self.receiveData()
  File "steppingmotor/Trinamic_control6110.py", line 94, in receiveData
    RxBuffer = self.port.recv(9)
socket.timeout: timed out
Traceback (most recent call last):
  File "/kagra/apps/pcaspy/lib/python2.7/site-packages/pcaspy/cas.py", line 416, in read
    def read(self, *args): return _cas.PV_read(self, *args)
RuntimeError: SWIG director method error. Exception Calling Python Code
filename="../../../../src/cas/generic/st/casStreamOS.cc" line number=479
Internal failure - unexpected problem with client's input - forcing disconnect
/kagra/apps/pcaspy/lib/python2.7/site-packages/pcaspy/cas.py:235: RuntimeWarning: tp_compare didn't return -1 or -2 for exception
  if primitiveType in [aitEnumUint8, aitEnumInt8]:

vis-ikeda, 2020-09-29:
**使用方法の説明資料を作成**

#162 へまとめて記載

MiyoKouseki, 2020-10-14:
これはCloseします。

closed 2020-10-14 by MiyoKouseki