gw-vis/userapps #111: Thermal control

MiyoKouseki, 2020-09-10:
[From here](https://paper.dropbox.com/doc/Plan-for-the-long-brake-towards-O3c--A6BHN4gn~Ey5XkTLymY8BCZaAg-KqexeqMaCA7DITFGxA5ih)

gw-vis/kagra-vis#20 : make a prototype

AyakaAShoda, 2020-09-11:
Comments from VIS meeting 2020/9/11
* Having thermometers inside on the EQ stop?
* What is the target temperature? for room temperature, and for the suspension temperature.
(How much power required for the heater?)

MiyoKouseki, 2020-09-14:
- 最小限のプロトタイプ
 - SRに取り付けて温調
 - ヒーターの取り付け場所を試行錯誤
 - GASの制御信号が安定するかテスト

TypeAなら（IXかIY）
 - 中のセンサーを使って改修作業前にGASの信号をチェックする
 - クロス管に巻く
なら、できる。
 - プロトタイプを使う

vis-ikeda, 2020-10-28:
[リボンヒーター試験結果](https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12183)

MiyoKouseki, 2020-11-04:
@Ryutaro-T @vis-ikeda 温調システムの今後の予定はどうなりますか？

vis-ikeda, 2020-11-05:
結果から加熱してもすぐに反応はなかった。
[方針]
・クロス菅を常時温める方向で進める。
[日程確認]
・温度調整機材：12/16納期(メールから確認)
・IXの真空引き12/Mくらいを目指して進める。