gw-vis/userapps #132: ETMY commissioning

MiyoKouseki, 2020-09-12:


NaohisaSato, 2020-11-20:
Memorandum:
The arrangement of the BF Dumper in EX,EY is rotated about 5 degrees from the direction relative to the beam path. (see below Type-A report in the link ). This situation will be eliminated with this upgrade.
http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/VIS/VISMinutes20200515

NaohisaSato, 2020-11-20:
About BF Dumper coil numbering:
EY is numbering the coils (H1-H3, V1-V3) differently than elsewhere (EX, IX and IY). 
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=9838

It is possible to re-number them to match the others just by changing the pin placement in the cable.
Should we make this change at the time of upgrade?