gw-vis/userapps #142: Make test environment for Stepper motor control

MiyoKouseki, 2020-09-13:


MiyoKouseki, 2020-09-15:
三代様

天文台の佐藤です。
以前頼まれました、ステッピングモーター他を神岡に送る件ですが、
本日、下記の物品をゆうパックで送ります。

・モータードライバー
・モーター本体＋Yaw駆動機構試作品（※）
・リミットスイッチ実験セット
（※）駆動機構は素手では触らないようにしてください。
　　　（特にベローズとリミットスイッチ本体）
　　　分解と組立は下記を参照してください。

https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11598

大変遅くなってしまい、申し訳ありませんでした。
よろしくお願いいたします。

vis-ikeda, 2020-09-15:
#110 に詳細を記載

vis-ikeda, 2020-09-15:
TRINAMIC社のWindowsアプリケーションを用いてテスト環境を構築
・Driver for NPort 5100A Series (Windows Driver Manager (for Windows 7, Server 2008, or later))
・TMCL-IDE Version 3.0.25.0

[確認内容と結果]
・Direct modeを用いて
　1.L/R SW確認
     ・6.GAP get axis parameter
   　- LeftSw押し込み時に11.Left limit switch status: 1となること: OK
　   -RightSw押し込み時に12.Right limit switch status: 1となること：OK
　2.Reference Searchを用いてシステムが正常に動作していること : OK
     ・6.GAP get axis parameter - 193.Reference search mode:3に設定後、
        13.RFS reference search - 0.START start ref. search開始して正常に左右SWを検知することを確認

MiyoKouseki, 2020-09-15:
確認しました。Closeします。#110に進んでください。

closed 2020-09-15 by MiyoKouseki