gw-vis/userapps #149: How do we connect the relase sensor to the binary input.

MiyoKouseki, 2020-09-14:
How?

According to the document "Drawings No.20-30.pdf" in [JGW-D1809192](https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=9192), output signals of the release sensors named as EQ1,2,3,4,5 will connect to the binary input on the DGS rack not to the temporal indicator used in O3.

In this issue, we discuss how do we design the wiring design from the release sensor to the BI.

- [ ] Please upload the circuit diagram of the indicator used in O3.
- [ ] How do we input the signal to BI?  Are there any documents for BI card?
- [ ] Define the signal which send to BI.
- [ ] Design the wiring.
- [ ] Done.

NaohisaSato, 2020-09-18:
Touch sensor’s specifications:
[(タッチセンサ)AB9-1J-GN.pdf](https://github.com/gw-vis/userapps/files/5242833/AB9-1J-GN.pdf)
model number GN-PT5M3A for release sensor.

NaohisaSato, 2020-09-18:
Cartoon of the indicator box circuit
[Indicator box circuit.pdf](https://github.com/gw-vis/userapps/files/5242960/Indicator.box.circuit.pdf)
It's a very simple circuit using a constant-current diode.

MiyoKouseki, 2021-03-12:
https://dacgit.icrr.u-tokyo.ac.jp/issues/546?issue_count=37&issue_position=2&next_issue_id=541&prev_issue_id=547
に移動

closed 2021-03-12 by MiyoKouseki