gw-vis/userapps #155: Make Summarypage

MiyoKouseki, 2020-09-14:
## Is your feature request related to a problem? Please describe.
we have not prepare Summary Page. It is important to know the status of VIS channels quickly.

## Describe the solution you'd like
At least, we need GuardState summary for all suspension and ASD for TM to know if the motion can satisfy the requirement.

## To Do 
- [ ] Add GuardState summary 
- [ ] ASD of TM 
- [ ] others?
- [ ] Done

MiyoKouseki, 2020-10-12:
## Contents of SEI and SUS

### SEI
[image]

### SUS
[image]

MiyoKouseki, 2020-10-25:
## GuardianOverview

LHOのSUSのGuardianOverviewは[ここ](https://ldas-jobs.ligo-wa.caltech.edu/~detchar/summary/day/20201025/sus/guardian_overview/)。これをKAGRAに置き換える。対応関係は以下の通り。

### 参考
 - sus-guardian.ini 

### 基本的な対応関係
- SUS_ -> VIS_
- MC1 -> MCI
- MC2 -> MCE
- MC3 -> MCO

Guardianのチャンネルは結構違うとおもう。置換でなんとかなると思うけれど。

## Oplev overview

SUS:OpLev Overviewは[ここ](https://ldas-jobs.ligo-wa.caltech.edu/~detchar/summary/day/20201025/sus/oplev_overview/)。
まずはこのページにある、angular optic motion [OpLev] を、Type-A、BS,PR3,SR3についてつくる。

### 参考
 - sus-oplev.ini

### 対応関係
 - M3,L3はTMに
 - 

## SEI Summary
LHOのは[ここ](https://ldas-jobs.ligo-wa.caltech.edu/~detchar/summary/day/20201024/sei/)。

LIGOの場合、色んな種類の防振台があるが、KAGRAはPre−Isolatorだけ。
これは、上2つができたらやる。それまで保留。

### 参考
 - 

### 対応関係

yuzuri, 2020-10-26:
## 議論メモ(三代, 押野, 山本, 譲原)

- KAGRAのガーディアンはステップが多い、LIGO並にシンプルに
[https://ldas-jobs.ligo-wa.caltech.edu/~detchar/summary/day/20201025/sei/](https://ldas-jobs.ligo-wa.caltech.edu/~detchar/summary/day/20201025/sei/)

- KAGRAのSEI相当は、TypeAとTypeBの合計8つ
 - ここに目標感度(reference)を重ねがきしたい
- それぞれのpre-isolaterの加速度計のスペクトルを重ね書きする
 - 対応するチャンネルを三代くんが指定
 - referenceデータも三代くんによる提供
- ひとまず、すべてLIGO準拠で、独自に何かを追加しない方針
- `VIS / Pre-Isolater / ...` という段階構造
- SEI : SUSが載ってる油圧の防振系、RMSが変化するのはここ
- SUS : SEIの上に載ってるサスペンション
- KAGRAのサマリーページは`VIS / Pre-Isolater / BS, SR2, SR3, SRM, .....`
- ground motionはGIFのsectionにまとめる

- VISのガーディアンサマリーに円グラフを付けたい
- overviewだけ見せるのがLIGO方式、円グラフをあえて隠す設定にしてる
- `VIS / guardian / 各サスペンションの名前`を作る
- VIS suspension managerというページがあると良い(1年後に三代くんが作る)
- ガーディアンのstate一覧を表示するコマンド
 - `guardutil states VIS_ETMX`

- k1sum0にテスト用のディレクトリがある
- VISのini fileは今3つに別れてる
- k1vis.ini          : これを触る
- k1vis-oplev.ini    : 触らなくていい
- k1vis-guardian.ini : 各タブにhidden == trueを消せば円グラフが出てくると思う

MiyoKouseki, 2021-01-15:
これってどうなっていましたっけ？

yuzuri, 2021-01-15:
完全に忘れてました・・・
週末に取り掛かります

yuzuri, 2021-01-15:
前にDetChar meetingで、VIS関連のリアルタイムモデルが修正されてると聞きました
なので、チャンネル名もまだ最終版ではないかもと聞いたのですがどうでしょうか？

yuzuri, 2021-01-15:
前にDetChar meetingで、VIS関連のリアルタイムモデルが修正されてると聞きました
なので、チャンネル名もまだ最終版ではないかもと聞いたのですが作業を始めちゃっても良いでしょうか？