gw-vis/userapps #157: Why do we need the offload?

MiyoKouseki, 2020-09-15:
そもそもなぜオフロードしないとだめなのか？そもそもメカの改良をすべきではないか？

例えば、一応、メカの改良案は高橋さんも考え始めている。
```
皆さんから出ているテーマ以外ではIPの低周波数化というのがあります。
IPは現在70mHz程度にチューニングされていますが、これを30mHzまで下げます。
その場合DC制御が入っていないと容易に倒れてしまうので、リモートで
機械的にロックできる機構が必要となります。その機構の開発です。
9/9のメールより
```

NaohisaSato, 2020-09-15:
すいません
何か間違いで removed their assignment になってしまいました

MiyoKouseki, 2020-09-15:
アサインし直してもよいですか？
今後IPを低周波数化して不安定になった場合、それを機械的にロックする機構の開発が必要になります。
そのときに佐藤さんのお力が必要になるので。。

NaohisaSato, 2020-09-15:
再登録をお願いします。何か押してしまったようでして

MiyoKouseki, 2020-09-24:
現状の洗い出し

- [ ] 
- [ ] 
- [ ]

NaoatsuHirata, 2020-09-30:
ステッピングモーターがついているバネの拡大図、IPのストッパー部について、作図しました。確認をお願い致します。
https://1drv.ms/u/s!ArKdG0skeXd_jO4KAWcRsc_oKPkS_Q?e=nrRvVr

NaoatsuHirata, 2020-09-30:
図を追加してアップデートしました。最新版は”Blade spring with stepper motor for IP_200930”です。
https://1drv.ms/u/s!ArKdG0skeXd_jO4KAWcRsc_oKPkS_Q?e=nrRvVr