gw-vis/userapps #160: Procedure document during the venting

MiyoKouseki, 2020-09-16:


MiyoKouseki, 2020-09-16:
Hello all,

Upon request I wrote a brief but descriptive procedure about how to take care of the suspensions during venting.

It does not replace the actual demonstration which I will be giving you in due time, but it will be good to read it before hand.

See section 4.1 on page 9 in the document at

https://1drv.ms/b/s!AoQaV1tdsEcUgdBD4tVVUon9mmf7Zw?e=pZncoj

Please let me know of any comments you may have.

Thank you in advance.

Regards,

Fabian

MiyoKouseki, 2020-09-16:
@naibafomsare (Fabian), Thank you so much. I read Chapter 4.1.

1. About the 1st item in the procedure,
```
Takes  notes  of  the  positions  of  all  GAS  Filter  keystones  (12 values  in  Type-B)  before  the  venting procedure beg
```
Could you write the channel name for taking notes of the position? 

2. Have we checked the direction of the stepper motor for L, T, and Y in all suspensions having the IP? I'm afraid that I will be in a panic if the stepper will not work properly. 

3. About the 8a item,
```
The usual number of steps required is of the order of a few thousand, but not much more  than  5,000  steps.  Experiment  with  fewer  than  5,000  steps  first  until  you  feel  comfortable with a reasonable value.  
```
Can you say the steps's value as the displacement?

4. Thank you!

naibafomsare, 2020-09-17:
Items 2 and 3 don't have a pretty answer. I'll explain in the document.

MiyoKouseki, 2020-09-25:
@naibafomsare Do you think you have finished writing a procedure for the venting part?
I'm going to close this issue if you think that.

MiyoKouseki, 2020-10-15:
Is it dangerous to move the stepper motor while the FB control is engaging?

naibafomsare, 2020-10-16:
> 
> 
> Is it dangerous to move the stepper motor while the FB control is engaging?

According to our experience it is safe for both, the IP and GAS filters. The control system is robust enough to allow us to to this. As far as I know, the only time this procedure failed was because a stepper motor driver broke and the DAC saturated.

It is fine to do it with the control system off also. I think Lucia preferred it that way.

MiyoKouseki, 2020-10-16:
The way to reduce the control signal is more convenient when doing OSEM Offload in Picomotor. Because I think it's better to offload OSEM's DC value when it's aligned.

naibafomsare, 2020-10-16:
> 
> 
> The way to reduce the control signal is more convenient when doing OSEM Offload in Picomotor. Because I think it's better to offload OSEM's DC value when it's aligned by some feed back control.
> 
> By the way, Ikeda-san is making the OFFLOAD_OVERVIEW.adl for offload work with stepper and picomotor.
> https://app.zenhub.com/workspaces/vis-work-5f56e83aff8ec50012d90116/issues/gw-vis/userapps/178

- It is indeed convenient to offload the feedback actuation to fishing rods. 

- The IM control system should not be used during venting. 

- I'm looking forward to see the new interface. 👍

MiyoKouseki, 2020-10-16:
>The IM control system should not be used during venting.

Of course, we don't use OSEM control during the venting. So I write "by the way". The control should be when the IFO commissioning.
Thanks.

closed 2020-10-18 by MiyoKouseki