gw-vis/userapps #162: Write a manual for the limit switch installation

MiyoKouseki, 2020-09-24:
インストールマニュアルを書く。

https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11520

- [ ] Hardware 部分のインストールマニュアルを書く
- [ ] Software 部分のインストールマニュアルを書く
- [ ] インストール後の動作確認項目とその確認方法を書く
- Done

vis-ikeda, 2020-09-29:
**Software 部分のインストールマニュアルを書く**
Limit switchの確認方法
ソフトウエアインストール手順（TBD）も併せて記載
正式版はハードウエアのインストール手順と相談してから決める予定
[Limit switch connecting guide_rev.2.pptx](https://github.com/gw-vis/userapps/files/5296342/Limit.switch.connecting.guide_rev.2.pptx)

vis-ikeda, 2020-09-29:
**どこに何をインストールするか、情報を共有。**

satoさん、takahashiさん＞
　ハードウエアのインストール日程と順番について情報を共有したい。
　現状リミットスイッチの対応について
　ソフトウエアとしてはGASは試験環境で確認しているが、
　IP側は未確認なので最初はソフトウエアの試験を兼ねてのインストール作業になるものと考えております。
　その為、VISのインストールの日程について認識をあせておきたいと考えております。

MiyoKouseki, 2020-09-29:
@vis-ikeda @NaohisaSato @Ryutaro-T @で通知を飛ばせるので是非。

vis-ikeda, 2020-10-12:
・インストール後の動作確認項目とその確認方法を書く
#164にて記載

vis-ikeda, 2020-10-12:
・どこに何をインストールするか、情報を共有。
　今回Limit SwitchがInstallされるのはType-A, Type-BのIPより上が対象
　Type-Aは既にInstallされているものもあるがソフトウエアからみて同じなので問題なし。
　L-Rが入れ替わっている可能性があるのでチェックが必要とのこと。

MiyoKouseki, 2020-10-22:
もう、準備完了ですか？

vis-ikeda, 2020-10-26:
ソフトウエアはインストール方法が変更になったので自動でできない(LVDTを見ながらStepperを動かしてstepperの上限下限位置を探す)
MEDM追加と書き直しが必要な認識。

vis-ikeda, 2020-10-29:
Stepper用のMEDM [OVERVIEW]を新規作成
sitemapの[STEPPING MOTOR]から呼び出せるように変更、
今までの[STEP_MASTER]と[OVERVIEW]を選べるように変更しました。

これに伴いSoftware Installマニュアルを書き換えました。
[Limit switch connecting guide.pptx](https://github.com/gw-vis/userapps/files/5456543/Limit.switch.connecting.guide.pptx)

MiyoKouseki, 2020-11-04:
ありがとうございます。
Reviewに移動できますか？

vis-ikeda, 2020-11-04:
@NaohisaSato さんがまだ修正するとのことです。

NaohisaSato, 2020-11-05:
@MiyoKouseki , @vis-ikeda 修正終わりました。

MiyoKouseki, 2021-01-15:
ありがとうございます。Closeします。

closed 2021-01-15 by MiyoKouseki