gw-vis/userapps #163: Endurance test for limit switch

MiyoKouseki, 2020-09-25:
試作機のリミットスイッチがスタックして動かない問題が発生した。本番用の量産型ではそれが起きないことを確実に確認したい。そのための耐久テストを実施する。

- [ ] 試作機で起きた状況を整理
- [ ] 耐久テストのやり方をきめる
- [ ] 耐久テスト用のPython scriptを書く
- [ ] テストを実施して評価
- Done

予定としては、30日の午後に佐藤さんがきて量産型に取り替えてもらうので、その夜に耐久テストができる状態にしておく。なので、とりあえずは試作機で耐久テストができるように準備する。

MiyoKouseki, 2020-09-25:
試作機で起きた状況を整理

https://github.com/gw-vis/userapps/issues/110#issuecomment-698658809

vis-ikeda, 2020-09-25:
耐久テストのやり方をきめる

新規にテスト用のPythonコードを準備
[試験内容]
・モーターを:1000000(L側へ)へ駆動
・モーターの速度が0になるまで待機
・L/RリミットSWを読み取り(L:ON,R:OFF)
・モーターを-1000(R側へ)駆動
・モーターの速度が0になるまで待機
・L/RリミットSWを読み取り(L/R:OFF)→　あてつきから解除できているか確認
・クーリング期間(10sec)待ち
同様にR側でも同じ確認を行う
上記を100回繰り返して当て付き発生時に異常が発生していないかを確認

vis-ikeda, 2020-09-28:
**確認**
・メカ的に耐久試験を行っても良いか？
　→ 9/28朝MTG
　　　100回程度ならば大丈夫と思うが元々そんなに動作することを想定しておらず
　　　耐久性の観点では選定していない。

vis-ikeda, 2020-09-28:
**耐久テスト用のPython scriptを書く**

https://github.com/gw-vis/pcas-controller/tree/master/scripts/stepmotor/steppingmotor
/endurance_test_limitswitch.py

作成済み

vis-ikeda, 2020-09-28:
**テストを実施して評価**
・メカ的耐久性に不安がある。
・現状サンプルで0/50回で発生していない
・当て付け時の角度で発生するか決まる。
　直角に当たれば問題ない。発生時は、少し斜めにあたっていた。
→　上記を踏まえ9/28miyoさんと実施の有無、確認するのであればいつ実施するか再度相談

NaohisaSato, 2020-09-28:
コメント）Yaw 量産品の改良点
・Switch Leverの強化
・Switch LeverとCollar LS との摩擦抵抗軽減
・Collar LSの工作精度（なるべく軸が傾かないように、Leverのバタつき防止）
（[Yaw LS 図面](https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=11477)）

NaohisaSato, 2020-09-28:
Vertical の方は耐久試験はおろか試運転もできていません。
やるとすれば現場で真空槽が開いてからになります。

vis-ikeda, 2020-10-12:
耐久性が確保されていないので試験を実施するべきか悩ましい。
現状２０回ほど初期化でLーR間往復させているが問題は発生していない。

vis-ikeda, 2020-10-29:
次の理由で耐久試験は実施せずこちらもクローズしたいと考えております。
・メカの耐久性がない。
・通常時にリミットスイッチまで到達させないソフトウエア設計とした。

MiyoKouseki, 2021-01-15:
We don't do the test because we use the limit switch as a sensor to know the limit range of the stepper motor. so we don't touch the motor to the switch frequently.

closed 2021-01-15 by MiyoKouseki