gw-vis/userapps #164: Make EPICS channel to monitor the step value of the limit switch position

MiyoKouseki, 2020-09-25:
リミットスイッチを入れたので、LとRそれぞれのスイッチの場所をステップ数で取得できるようにする。これをEPICSに流せるようにすれば、ソフト上で間違ったステップ数を指定することを防ぐことができる。

ちなみにステッピングモーターは動かさないときには電源をOFFにするので、LとRのステップ数は不揮発性メモリに保存しておく必要がある。さらに言えば、Home位置はおそらくインストール時の位置で初期化して不揮発性メモリに保存していないと思うので、Home位置を初期化する方法を調べておかないといけない。

インストール完了時には初期化の手順を行わなければならない。まず、インストール完了したその位置でHome位置を保存する。つぎにモーターを動かしてLとRの場所のステップ数を保存させる。この初期化作業が必要。

**ToDo**
- [ ] L,Rのスイッチが何ステップのところにあるか取得する関数をつくる
- [ ] Homeポジションの初期化手順をしらべる
- [ ] 取得したステップ数をドライバの不揮発性メモリに記録（Home,L,Rについてそれぞれ）
- [ ] L,Rのステップ数用のチャンネルをつくる
- [ ] 初期化手順をインストールマニュアルに書く

vis-ikeda, 2020-09-28:
[調査結果]
・モータードライバ側の現在位置は電源をOFFにするとモーターがどこにいても現状0からスタートする。
・TMCM-6110_TMCL_firmware_manual_Fw1.42_Rev1.17.pdf
　P.76
　　4 Axis Parameters
  　　1.Actual position:現在位置(Lからの距離として使用)　=　インストール時の位置とする
         196:End switch distance:L-R間の距離　＝ インストール時の位置からL、Rに駆動させて距離を測定する
　　3.7.13 RFS (Reference Search)が使えるか要検討、使えなければ検出用PGを作成

vis-ikeda, 2020-09-28:
satoさんへ要確認
　L/RのLimit Switchの位置まで駆動させて[Lswitch]- [Home(取り付け時の最初の位置) -[Rswitch]間の距離を
　測定してユーザーにそれ以上の値を入力させないようにブロックすることを考えている。
　そもそもLimit switchの取り付けられている位置まで駆動させても問題ないでしょうか？

NaohisaSato, 2020-09-28:
Limit switchが働いていれば、Limit switchの取り付けられている位置まで駆動させても当然大丈夫です。
働いていない場合も、スイッチのある位置まで（＋さらに1~2mm）なら大丈夫です。

vis-ikeda, 2020-10-02:
### Design
**不揮発への保存内容**
[Home位置]、[LーR距離]、停電に備えて[現在の位置]を不揮発に記録する。
不揮発は"User Variables in Bank 2"の最初の56個が不揮発で起動時に自動的にRAMへ展開されるのでここへ書き込む。
モーターがCH0から5まであるので不揮発への保存は以下の並びで保存する。

```
Bank.2
　Number.0-5: [Home位置]CH.0から5
　Number.6-11: [LーR距離]CH.0から5
　Number.11-16: [現在の位置]CH.0から5
　Number.17-55:[Reserve] 

```
→ 月曜日にここへの書き込みで要件が満たせるか試験を実施する予定。

・MEDMから[INIT]ボタン押下でk1scriptで上記距離を
　[Home位置]→[L limit switch]→[R limit switch]へ移動して
　測定、位置とサイズを不揮発へ保存する。
　初期化後は[INIT]ボタン押下前の[Home位置]へ復帰して終了。
・初期化は、モーター1CH毎に行う。IPも1Ch毎行うこととする。

以降は、[POSITION]の範囲は、測定された[LーR距離]の９０％を上限(中心はLーR距離の半分で両側９０％)
のみ駆動できる領域としてリミットスイッチに当てないものとする。
電源OFF時、次回電源ONで読み込まれた座標で初期化を行うことで状態を復帰させる。

vis-ikeda, 2020-10-08:
2020/10/9「リミットスイッチ レビュー　10/9」説明用の資料

[Stepper Motor Limit Switch Software Design.pptx](https://github.com/gw-vis/userapps/files/5344820/Stepper.Motor.Limit.Switch.Software.Design.pptx)

NaohisaSato, 2020-10-08:
池田さん
試作機の絵を添付しました。良ければレビュー資料の2ページ目に使ってください。

[image]

MiyoKouseki, 2020-10-08:
【コメント】
### Limit Switch の位置
これはもしかしたら池田さんというよりは @NaohisaSato  さんへのコメントになるかもしれません。昨日高橋さんと話をしたのでそれをまとめます。

基本的には、モーターがスタックしないような位置（佐藤さんが説明？）にリミットスイッチを置く。が、ユーザーが大きく操作できないように、制限する箇所もある、かもしれない。

 - Vertical : 基本の位置にする※2
 - IP ： 基本の位置にする。※2
 - Yaw : ステッピングモーターの軸上で、±1mmの位置に置いて制限する？※1

※2 この場合、振り子がどこかに当たるので、それを検知できずに動かし続ける可能性がある。そういうことを考えると、やはりLVDTの信号を常にみて、これが線形なレンジに収まるようにAuto-offloading を作る、ということになる。
※1 これはYawに計算したときにどれぐらいの角度になるか、計算をすべき。高橋さん曰くすでにそれはしてあるそうなので、それを別のIssueとして載せる。LVDTでみるならこの制限はいらない気もする。制限しないほうが良い気がする。

### 全体の説明
@vis-ikeda さん、Apendixとして、Stepperがインストールされている場所の絵を載せてもらえますか？
https://github.com/gw-vis/kagra-vis/tree/master/SuspensionControl

そしてStepperの対になるセンサーをまとめた以下の表もスクショを貼り付けてもらえませんか？

| Stepper | Sensor | Comment | 
|:---|:---|:--- |
| IP_{L,T,Y} | LVDT_{L,T,Y} | Tye-A,B |
| {F0,F1,F2,F3,BF}_GAS | F0_GAS | Type-A,B |
| F0_YAW | BF_LVDT_YAW | only Type-A |
| F0_YAW | No? OPLEV_YAW? | Type-B, (Bp?) |

Picoについて
|Picomotor | Sensor | Comment |
|:---|:---|:---|
|BF_YAW| NO? OPLEV_YAW?| Type-A,B |
|BF_{PIT,YAW}| BF_LVDT_{PIT,YAW}| TypeB, (A?) |
|IM_{PIT,ROL}| OSEM_{PIT,ROL} | Type-B,Bp |

vis-ikeda, 2020-10-08:
@MiyoKouseki さん、Overview.ppt.へ貼り付ければ良いですか？

MiyoKouseki, 2020-10-09:
@vis-ikeda いえ、池田さんのスライドに載せていただけますか。

vis-ikeda, 2020-10-09:
@MiyoKouseki @NaohisaSato 頂いた画像、表を資料へ反映しました。

[Stepper Motor Limit Switch Software Design.pptx](https://github.com/gw-vis/userapps/files/5351917/Stepper.Motor.Limit.Switch.Software.Design.pptx)

vis-ikeda, 2020-10-12:
・L,Rのスイッチが何ステップのところにあるか取得する関数をつくる
・Homeポジションの初期化手順をしらべる
・取得したステップ数をドライバの不揮発性メモリに記録（Home,L,Rについてそれぞれ）
→ stepper_install.pyにて実装
　Homeの他現在位置も用意。Homeは作成したが使い道が今のところない。

・L,Rのステップ数用のチャンネルをつくる
→_LIMITPOSMIN,_LIMITPOSMAXとした

 ・初期化手順をインストールマニュアルに書く
[Uploading Limit switch connecting guide_rev.2.pptx…]()

MiyoKouseki, 2020-10-14:
He wrote some setting on the stepper motor driver so that we can store the position of the limit switch.
He also prepare some EPICS channel for the position.

We reviewed this work on 10/09.

I close.

closed 2020-10-14 by MiyoKouseki