gw-vis/userapps #166: model selection

vis-ikeda, 2020-09-30:


vis-ikeda, 2020-09-30:
[機種選定検討](http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/VIS/TempControl)

vis-ikeda, 2020-10-01:
10/1 愛三電機株式会社 永井様へ見積もり依頼中

内容
・Panasonic KT4R(AKT4R2131001) ：２台
　操作電圧：AC/DC24V、直流電流出力、通信機能ありのもの
・Panasonic KS1 信号変換器 ：１台
・RS485ケーブル（3線シールド付きツイストペア）: 1m x 2本
　上記機材の通信に使うので
　平河ヒューテックさんのコンピュータ用電装ケーブル
　かこれ相当のもの

vis-ikeda, 2020-10-07:
>・Panasonic KT4R(AKT4R2131001)　２個
>　⇒最短12/16納品予定
>　※完全受注生産品のため、どうしても納期が長くなってしまうとのことです。
>
>・Panasonic KS1(AKS1202) 信号変換器　１個
>・計装用電線(HAVVS1806)　10M
>　⇒この2点は最短で10/9に納品できますがいかがいたしましょうか。
>　　一括納品御希望でしたら12/16にまとめて納品させていただきます。

→　一括納品

MiyoKouseki, 2020-10-14:
He selected the model of the temperature controller.
At first, he had ordered some stuff to vender to make prototype for the control system.

Now we waiting for delivery at 12/16.

Close

closed 2020-10-14 by MiyoKouseki