gw-vis/userapps #178: Make OFFLOAD_OVEVIEW.adl

MiyoKouseki, 2020-10-11:


MiyoKouseki, 2020-10-11:
I want to have an overview screen for the offloading so that we can move the stepper motor by viewing the oplev like graph or indication bar.

For example, The screen is like the following.
[image]

MiyoKouseki, 2020-10-11:
Please add the EPICS channel for the guardian status like K1:GRD-VIS_ETMX_STATUS_S to confirm that the suspension is controlled only at the IP or GAS stages during the offloading.

vis-ikeda, 2020-10-13:
・ステップ数を変えられるようにする。
・ndscopeを呼び出せるようにする。
・guardianのstateを表示する。

vis-ikeda, 2020-10-13:
試作画面

[image]

MiyoKouseki, 2020-10-13:
（さきほど口頭でお伝えしたことをメモしておきます）

ありがとうございます！
自分でお願いして申し訳ないのですが、やはりIPのLenとTraを二次元プロットするのではなく、YAWのように横向きのバーにしてもらえますか？また、GASも縦向きではなく、同様に横向きにしてもらえますか？つまりすべてが横向きのバーに統一する感じです。

すべて横向きにすれば、「緑位置から右にずれたら左ボタンを押す」という具合に簡単に説明できて、説明を受ける側も理解しやすいなと思いました。さらに言えば、そもそもこの画面で「GASのキーストーンが上下だから縦」という情報は不必要だなと思ったので。

vis-ikeda, 2020-10-14:
試作二号

[image]

vis-ikeda, 2020-10-14:
現状モデルで不足しているチャンネル達

・Type-B, Type-Bpには、Type-AのSUMOUT(ex.K1:VIS-ETMX_F0_SUMOUT_L_OUT等)がなくバー表示に持ちいれるデータがないのでmodel修正時に組み入れて頂く必要がある。

MiyoKouseki, 2020-10-14:
- [ ] Find channel names for TypeB and TypeBp after the power outage

vis-ikeda, 2020-10-15:
・LimitSwitchの上限下限をユーザーが入力できるようにする
・sitemapへ追加

vis-ikeda, 2020-10-16:
memo
ふっと考えていて気になったので忘れないように記録しておく。
funcIP.py
Move_L()
        driver.setTargetPosition(posA+int(signA*value),motorA)
        driver.setTargetPosition(posB-int(1/2*signB*value),motorB)
        driver.setTargetPosition(posC-int(1/2*signC*value),motorC)

motorAの移動量が１とした場合
motorB,Cの移動量が1/2となる
このとき２つの問題がある。

問題点１
　1/2の計算はpython2の場合値が0になる。python3の場合0.5になる
問題点２
　問題点１を1.0/2.0に修正して移動量が意図通りとなったとして、
  motorAとB/CのそれぞれのL方向の物理的移動量がmotorA側が1に対して
  motorB/C側がmotorの駆動量は2等辺三角形で考えると
　モーターの移動量：実際のＬ方向の移動量＝2：1になるので
　物理的移動量は1/2でモーターも1/2しか駆動しないのであれば
　実際の移動量はmotorA側に対して1/4しか移動しないのではないかと考えている。
　motorAと同じだけ移動するのならばモーターの駆動量を2.0/1.0にして
　2/1*1/2＝1.0にしなくてはいけないのではないか？
　他Ｔ方向も同様に分母、分子入れ替えが必要？
　要確認。
        driver.setTargetPosition(posA+int(signA*value),motorA)
        driver.setTargetPosition(posB-int(2.0/1.0*signB*value),motorB)
        driver.setTargetPosition(posC-int(2.0/1.0*signC*value),motorC)

MiyoKouseki, 2020-10-17:
@vis-ikeda @NaohisaSato @NaoatsuHirata 問題2に関してですが、佐藤さんと平田さんと一緒に考えてもらえませんか？
L以外にもTとYについても。

vis-ikeda, 2020-10-17:
@NaohisaSato @NaoatsuHirata 

http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/DGS/Projects/StepperMotor
とりあえず昼から必要になるので上記記載のType-TMのいモーター駆動量について教えてください。

・まずこの図のstepperの位置がただしいか確認したい。

そのうえで以下、現状が正しいか教えてください。

・Long方向にN[step]駆動させる場合
　（モーターN[step]分の距離を移動すると仮定した場合の各モーターの駆動ステップ数を考えます。）
　現状: A :N[step], B,C: -1 / 2ｘN[step]
　推測: A: N[step], B,C: -2 / 1 x N[step]
・Trans方向にN[step]駆動
　現状: A :0[step], B:-sqrt(3) / 2 x N[step], C: sqrt(3) / 2ｘN[step]
　推測: A: 0[step], B:-2 / sqrt(3) x N[step], C: 2 / sqrt(3)ｘN[step]
・Yar方向にN[step]駆動
　Yar方向は指示値のまままわしていたので無関係でした。
　現状: A:N[step], B:N[step], C: N[step]

・Long方向は推測の様に分子分母を入れ替えないとAとB/Cの移動量が異なり回転してしまうか？
・Trans方向は分子分母を入れ替えなくても移動方向(距離が異なるだけで角度は同じ)
　は問題がないので影響がなさそう。
・Type-SR、BSは方向が変わるだけなので上記が決まれば同じと考えているが問題ないか。
　Type-SR
　・Long方向にN[step]駆動
　　現状: A: sqrt(3) / 2 x N[step], B: -sqrt(3) /2ｘN[step], C: 0[step]
　　推測: A: 2 x sqrt(3) x N[step], B: -sqrt(3) /2ｘN[step], C: 0[step]
　・Trans方向にN[step]駆動
　　現状: A: 1 / 2 x N[step], B: 1 / 2ｘN[step], C: -N[step]
　　推測: A: 2 / 1 x N[step], B: 2 / 1ｘN[step], C: -N[step]
　・Yar方向にN[step]駆動
　　Yar方向は指示値のまままわしていたので無関係でした。
　　現状: A:N[step], B:N[step], C: N[step]

　Type-BS
　・Long方向にN[step]駆動
　　現状: A: sqrt(3) / 2 x N[step], B: -sqrt(3) / 2 ｘ N[step], C: 0[step]
　　　　 → 問題３：実装されている計算式が図と異なる？！
　　推測: A: 2 x sqrt(3) x N[step], B: 0step]. C: -2 / sqrt(3) ｘ N[step]
　・Trans方向にN[step]駆動
　　現状: A: -1 / 2 x N[step], B: N[step], C: -1 /  2ｘN[step]
　　推測: A: - 2 / 2 x N[step], B: N[step], C: -2 / 1ｘN[step]
　・Yar方向にN[step]駆動
　　Yar方向は指示値のまままわしていたので無関係でした。
　　現状: A:N[step], B:N[step], C: N[step]

Type-TMのみ10/19 午前中に回答をください。

NaoatsuHirata, 2020-10-19:
- SRについて
下記リンク先アセンブリ図10ページ目にSteppingmotor等の位置を記しています。
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=7077
これはSR3ですが、デザインはSR2とSRMも同様です。

vis-ikeda, 2020-10-19:
修正前の状態で良いと言うことなので問題点１のみ修正して他はそのままとした。

NaoatsuHirata, 2020-10-20:
- BSについて
下記リンク先アセンブリ図の10ページ目にSteppingmotorの位置を記してあります。
https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=5092

vis-ikeda, 2020-10-20:
Design変更
- [ ] IPの表示はLEN,TRA,YAWにしているが、 BS以外は「センサーの取り付け位置」と「stepper motorの取り付け位置」が同じなのでLEN,TRA,YAWの表示からH1,H2,H3に変更する。
- [ ] BSについては「センサーの取り付け位置」と「stepper motorの取り付け位置」が異なるのでどうするか @MiyoKouseki さん預かりとする

vis-ikeda, 2020-10-20:
Design Change LTY to H1,H2,H3 (BS no change)

[image]

vis-ikeda, 2020-10-20:
- [ ] script(step_start)の開始終了が画面上から行えるようにすること

vis-ikeda, 2020-10-22:
- [] 表示範囲の上限下限のEPICSチャンネルを追加して手動で変更できるようにする

MiyoKouseki, 2020-10-22:
@vis-ikeda とりあえず、一通り動くものができたので、一旦このIssueをCloseしたいと思います。
残りのToDoは他のIssueとしてAuto-offloading Systemの子Issueとして作成してもらえますか？

MiyoKouseki, 2021-01-15:
Close. now we can access the OFFLOAD_OVERVIEW.adl from sitemap.

closed 2021-01-15 by MiyoKouseki