gw-vis/userapps #182: know wether the FR touching on mechanical structure by reading the motor torque

MiyoKouseki, 2020-10-17:


vis-ikeda, 2020-11-24:
・#195でLOADの値読み込みを検討中。
・モータードライバ側にはstallGuard2という負荷値がある閾値を超えた場合に停止することができるシステムがある。
　ただし単純なパラメータだけではないので導入にはモータードライバ側の組み込みソフトを少し作成する必要があり、また負荷値の閾値を決める必要がある。負荷値によってはモーターが駆動しなくなるので細心の注意が必要。

vis-ikeda, 2020-11-30:
[stallGuard2](http://gwwiki.icrr.u-tokyo.ac.jp/JGWwiki/KAGRA/Subgroups/DGS/Projects/StepperMotor/stallGuard2)
へコメントをまとめた。

LOAD値は0で負荷が高くなる。
これを使用するためには２つパラメータを決める必要がある。
{{{
SAP 174, 0, 2    //stallGuard2 threshold
→stallGuard2の感度は0から+1増やして(感度を下げて)LOADが0になる箇所を見つけて設定
SAP 181, 0, 0    //smartEnergy stall velocity
→velocityは今回は0以上でstallGuardが反応するように設定
}}}
感度は計算で算出できないので正しく動作している時に実測する必要がある。
感度が高くても低くてもLOADの値が０（高負荷状態）にならない。（まだこの値を決める経験が少ないせいもある）
電流変化から測定していると思われるのでLOADによる負荷測定はあまり精度の高い検出方法ではないと考えている。
現状動いているモーターも測定の為、長い距離動かすことが難しいので測定は難しいのではないかと考えている。
感度を変化させながら何度かモーターを往復させて一定速度の状態になった際に一番感度が高い値でLOADが一定の値になる測定を決める必要がある。（感度が高いとLOADがすぐに０になる。感度が低いとストールしても感度が０にならない。）
その為、導入は難しいと判断しております。

MiyoKouseki, 2021-01-15:
> その為、導入は難しいと判断しております。
了解しました。検討ありがとうございます。

closed 2021-01-15 by MiyoKouseki