gw-vis/userapps #188: Creating Gurdian Class and Package Diagrams

vis-ikeda, 2020-10-30:
[ClassDiagram](https://app.zenhub.com/files/276073075/b570aa1d-80f3-4d45-a1de-c0dac428d36c/download)
[package Diagram](https://app.zenhub.com/files/276073075/cf11f5ff-5402-4086-a5a5-be5436e7b4a1/download)
未使用ファイルを消していないので少し大きい・・・

[Usage]
 python2.7 & pip & pylint(pyreverse) & Graphvizをインストール
 guardianのcommonとK1フォルダを1つのフォルダへコピー
 pyreverse -ASmy -f ALL -o dot *.py
 .dotをVsCodeからGraphviz Interactiv Previeで開いてブラウザで起動。
 ブラウザでpdfで保存した

vis-ikeda, 2020-11-02:
[Package Diagram TypeBp](https://app.zenhub.com/files/276073075/0767c31f-6135-481c-8069-3761aef1221e/download)
[Class Diagram TypeBp](https://app.zenhub.com/files/276073075/6b61d271-8f8f-4523-a3fb-f81f6b5b3905/download)
もっと詳しいのを作成するにはエイリアス先のファイルが必要だが一旦ここまでとします。

vis-ikeda, 2020-11-02:
edgesの定義をDOT言語化するpythonファイルを作成して現状のedges定義を確認。

[GRD_OLD.py : TypeBp](https://app.zenhub.com/files/276073075/fdf773f4-c071-42ed-a163-8e722054fb39/download)

vis-ikeda, 2020-11-10:
dev-prequa版:
[GRD_VIS.py : edges](https://app.zenhub.com/files/276073075/4ade17a9-76f0-4845-a346-7724ed768af4/download)
[Package Diagram All](https://app.zenhub.com/files/276073075/fb1c4567-4904-47c1-8f83-00096be3a5ed/download)
[Class Diagram All](https://app.zenhub.com/files/276073075/9c42e019-1242-41a8-9e7f-85d5effd1368/download)

vis-ikeda, 2020-12-09:
学習が目的でしたのでcloseします。

MiyoKouseki, 2021-01-15:
This issue was for Ikeda-san. He learn structure of the VIS guardian by reading actual code.

closed 2021-01-15 by MiyoKouseki