gw-vis/userapps #189: Check the guardian source code 'class check_TWWD()' with the function name pre_exc()

vis-ikeda, 2020-11-02:
'class check_TWWD()'の関数名pre_exc()は全ファイルで統一されている？
他の関数ではpre_exec()になっている。
変更不要？使用していないのならば削除。要確認。

以下、関係しそうなドキュメント
https://dcc.ligo.org/public/0111/G1400016/007/guardian-introduction.pdf
P.43
Guardian includes a special GuardStateDecorator class specifically designed for wrapping state methods:

https://dcc.ligo.org/public/0120/T1500292/001/AdvancedLIGOGuardian.pdf
P.35
The GuardStateDecorator pre_exec is set to be the NodeManager.check_fault method. The option “fail_return” argument is a return value for the decorator in case the check fails (i.e. a jump state name) (default None).

vis-ikeda, 2020-11-12:
/kagra/apps/guardian/lib/python2.7/site-packages/guardian/state.py
class GuardStateDecorator(object):
    def __call__(self, state_obj, *args, **kargs):
        pre_exec_return = self.pre_exec()
       ....
GuardStateDecoratorではクラスインスタンスの呼び出し時に__call__からself.pre_exec()を呼び出している。
このクラスを継承するとself.pre_exec()の呼び出しは、継承した側のクラスにあるpre_ecec()が呼び出される。
継承した側に関数がpre_exec()ない場合は、継承元GuardStateDecoratorにある空の関数が呼び出されるので関数名pre_exc()はpre_exec()に修正する必要があると思われます。

MiyoKouseki, 2021-01-15:
Guardianは最初から書き直すので、既存のコードの変更はしないということにします。
今後書くコードでは注意が必要です。

closed 2021-01-15 by MiyoKouseki