gw-vis/userapps #195: Operation SRM Rescue

vis-ikeda, 2020-11-16:
K-log [#15447](http://klog.icrr.u-tokyo.ac.jp/osl/?r=15447)

Fabian's request.
He asked me to incorporate it into the python code so he could test it next.
 - [ ] High Speed.(Now 10 or 500 to 1000 or Max.2000)
 - [ ] Both directions (Try moving it repeatedly in both directions.)
 - [ ] Show on the screen the torque. (But it's very difficult.)
 - [ ] Change and display to FULL(HALF) step. (Now it's 256 microsteps.)
We need to do this by 11/27.

naibafomsare, 2020-11-16:
[image]

vis-ikeda, 2020-11-16:
[memo]
Torque can be obtained by Actual Load Value (but you need to enable stallGuard2 to use it)
TMCM-6110 TMCL
```
 Number: 206
 Axis Parameter: Actual load value
 Description: Readout of the actual load value used for stall detection (stallGuard2).
 Range[Units]: 0...1023
```

vis-ikeda, 2020-11-17:
Specifications of the motor used
[D42.2-H](http://www.yasaka-tec.co.jp/images/vacuum/AML%20UHV%20Motor%20Jp.pdf)

If you look at the 'Step Frequency (Hz)', the torque is almost constant up to about 100 Hz regardless of the current value.

vis-ikeda, 2020-11-19:
> High Speed.(Now 10 or 500 to 1000 or Max.2000)
Just put the value in VEL.
However, since it works with about 15,000 [pps] even at 500, you might want to challenge the lower value. For example, 1 (=30[pps])

> Both directions (Try moving it repeatedly in both directions.)
I wrote some simple code in python to be used from MEDM.

> Show on the screen the torque. (But it's very difficult.)
Try to view the LOAD. Not sure if it works.

> Change and display to FULL(HALF) step. (Now it's 256 microsteps.)
Made selectable in step_start for rescue.

Written instructions for use.
[E2012245-v1](https://gwdoc.icrr.u-tokyo.ac.jp/cgi-bin/private/DocDB/ShowDocument?docid=12252)

@naibafomsare Will you try it?

MiyoKouseki, 2021-01-15:
Jamming of FR was no resolved by this way.

http://klog.icrr.u-tokyo.ac.jp/osl/?r=15479

closed 2021-01-15 by MiyoKouseki